CH3NGYZ/small-tailscale-openwrt #1: 手动安装 跑不起来

gzxiexl, 2025-04-16:
我按你之前教程里的手动安装
发现这个缩小版tailscale跑不起来 运行命令没有反应

CH3NGYZ, 2025-04-16:
我试试吧

gzxiexl, 2025-04-16:
我跑的是amd64的 我自己编译的也跑不起来 你的仓库里[Releases]下载的也跑不起来

CH3NGYZ, 2025-04-16:
我测试也无法跑起来,难道amd64不支持?

gzxiexl, 2025-04-16:
https://github.com/gzxiexl/glinet-tailscale-updater
你看下我Releases的 实测可以跑起来了 大小不到7M

[image]

你看下我改的流程文件吧 我也不知道什么原因 全靠deepseek一步一步指导

CH3NGYZ, 2025-04-16:
我这边也可以了,待我修改一下就行了

gzxiexl, 2025-04-16:
我建议脚本支持定期与官方tag比对 再判断要不要升级 reboot稍微有点影响

CH3NGYZ, 2025-04-16:
> 我建议脚本支持本地固化 定期与官方tag比对 再判断要不要升级 reboot稍微有点影响

有这个打算,到时候可选安装到本地还是每次开机装到tmp

gzxiexl, 2025-04-17:
我看了 amd64的都是7.1M 等你的脚本搞好再弄了😎

CH3NGYZ, 2025-04-17:
> 我看了 amd64的都是7.1M 等你的脚本搞好再弄了😎

可以试试了
安装没啥问题, 我还要完善卸载,自动更新以及其他脚本

gzxiexl, 2025-04-17:
下班回家试了下 我这网络连不上
```
root@iStoreOS:~# curl -o /tmp/pretest_mirrors.sh -L https://ghproxy.ch3ng.top/ht
tps://raw.githubusercontent.com/CH3NGYZ/ts-test/main/pretest_mirrors.sh && sh /t
mp/pretest_mirrors.sh
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  4223    0  4223    0     0   3918      0 --:--:--  0:00:01 --:--:--  5021
✅ 已更新镜像列表
测试 https://wget.la/https://github.com/ ...
/tmp/pretest_mirrors.sh: line 59: bc: not found
sh: invalid number ''
/tmp/pretest_mirrors.sh: line 60: bc: not found
✅ https://wget.la/https://github.com/  延迟: 0.00s  评分: 
测试 https://ghfast.top/https://github.com/ ...
/tmp/pretest_mirrors.sh: line 59: bc: not found
sh: invalid number ''
/tmp/pretest_mirrors.sh: line 60: bc: not found
✅ https://ghfast.top/https://github.com/  延迟: 0.00s  评分: 
测试 https://gh.catmak.name/https://github.com/ ...
/tmp/pretest_mirrors.sh: line 59: bc: not found
sh: invalid number ''
/tmp/pretest_mirrors.sh: line 60: bc: not found
✅ https://gh.catmak.name/https://github.com/  延迟: 0.00s  评分: 
测试 https://github.3x25.com/https://github.com/ ...
/tmp/pretest_mirrors.sh: line 59: bc: not found
sh: invalid number ''
/tmp/pretest_mirrors.sh: line 60: bc: not found
✅ https://github.3x25.com/https://github.com/  延迟: 0.00s  评分: 
测试 https://ghproxy.net/https://github.com/ ...
/tmp/pretest_mirrors.sh: line 59: bc: not found
sh: invalid number ''
/tmp/pretest_mirrors.sh: line 60: bc: not found
✅ https://ghproxy.net/https://github.com/  延迟: 0.00s  评分: 
测试 https://ghproxy.ch3ng.top/https://github.com/ ...
/tmp/pretest_mirrors.sh: line 59: bc: not found
sh: invalid number ''
/tmp/pretest_mirrors.sh: line 60: bc: not found
✅ https://ghproxy.ch3ng.top/https://github.com/  延迟: 0.00s  评分: 
✅ 最佳镜像: 
root@iStoreOS:~# ls
Configs
root@iStoreOS:~# curl -sSL https://ghproxy.ch3ng.top/https://raw.githubusercontent.com/CH3NGYZ/ts-test/main/
install.sh | sh
⬇️  尝试镜像: /CH3NGYZ/ts-test/raw/refs/heads/main/tailscale-openwrt-scripts.tar.gz
⬇️  尝试镜像: /CH3NGYZ/ts-test/raw/refs/heads/main/tailscale-openwrt-scripts.tar.gz
⬇️  尝试镜像: /CH3NGYZ/ts-test/raw/refs/heads/main/tailscale-openwrt-scripts.tar.gz
⬇️  尝试镜像: /CH3NGYZ/ts-test/raw/refs/heads/main/tailscale-openwrt-scripts.tar.gz
⬇️  尝试镜像: /CH3NGYZ/ts-test/raw/refs/heads/main/tailscale-openwrt-scripts.tar.gz
⬇️  尝试镜像: /CH3NGYZ/ts-test/raw/refs/heads/main/tailscale-openwrt-scripts.tar.gz
⬇️  尝试直连: CH3NGYZ/ts-test/raw/refs/heads/main/tailscale-openwrt-scripts.tar.gz
❌ 所有镜像与直连均失败，安装中止

```

CH3NGYZ, 2025-04-17:
> 下班回家试了下 我这网络连不上
>
> ```
> root@iStoreOS:~# curl -o /tmp/pretest_mirrors.sh -L https://ghproxy.ch3ng.top/ht
> tps://raw.githubusercontent.com/CH3NGYZ/ts-test/main/pretest_mirrors.sh && sh /t [...]

你这是因为系统没有安装bc, 我换了个方式实现, 你再试试?

CH3NGYZ, 2025-04-17:
用我readme里的最新链接试试

CH3NGYZ, 2025-04-17:
> 下班回家试了下 我这网络连不上
>
> ```
> root@iStoreOS:~# curl -o /tmp/pretest_mirrors.sh -L https://ghproxy.ch3ng.top/ht
> tps://raw.githubusercontent.com/CH3NGYZ/ts-test/main/pretest_mirrors.sh && sh /t [...]

因为你的安装前测速脚本pretest_mirrors.sh运行失败, 也就没有生成正确的valid_mirrors.txt, 应该是6行空格. 而后面install.sh需要这个文件的, 所以你运行不起来
[image]

CH3NGYZ, 2025-04-17:
我这边已经可以了

[image]

gzxiexl, 2025-04-17:
[image]
还是报错了

gzxiexl, 2025-04-17:
[image]3
还是报错了

CH3NGYZ, 2025-04-17:
怎么连jq都没有 👎🏻 , 我换了一个实现, 你再试

gzxiexl, 2025-04-17:
[image]
终于可以了

CH3NGYZ, 2025-04-17:
测试一下bark通知好使不

CH3NGYZ, 2025-04-17:
看起来还可以

[image]

gzxiexl, 2025-04-17:
我有一个建议
 ```
INFO: 🔧 管理自动更新：
[2025-04-17 20:47:48] 🔧 INFO:    /etc/tailscale/update_ctl.sh
[2025-04-17 20:47:48] 🔧 INFO: 
[2025-04-17 20:47:48] 🔧 INFO: 🔧 查看当前最新版本：
[2025-04-17 20:47:48] 🔧 INFO:    /etc/tailscale/fetch_and_install.sh --dry-run
[2025-04-17 20:47:48] 🔧 INFO: 
[2025-04-17 20:47:48] 🔧 INFO: 🔧 自动排序可用镜像代理：
[2025-04-17 20:47:48] 🔧 INFO:    /etc/tailscale/test_mirrors.sh
[2025-04-17 20:47:48] 🔧 INFO: 
[2025-04-17 20:47:48] 🔧 INFO: 🔧 管理server酱推送：
[2025-04-17 20:47:48] 🔧 INFO:    /etc/tailscale/notify_ctl.sh
[2025-04-17 20:47:48] 🔧 INFO: 
[2025-04-17 20:47:48] 🔧 INFO: 🔧 切换安装模式(重新安装)：
[2025-04-17 20:47:48] 🔧 INFO:    /etc/tailscale/setup.sh
```
像这样要用到的命令能否集成到一个命令，比如像shellcrash一样，输入shellcrash就出来各种菜单，比如TailHelper 搞个软连接放到/usr/bin下
直接就可以交互了，把若干下脚本集成到一起去，包括脚本本身自动更新。

CH3NGYZ, 2025-04-17:
脚本的更新直接就是安装前两步

CH3NGYZ, 2025-04-18:
写了个Tailscale helper, 可以一键进行一些操作, 其中脚本自身更新用9 更新脚本包就行

gzxiexl, 2025-04-18:
牛(òωó)👍 太肝了 这么快就搞出来了

closed 2025-04-18 by gzxiexl