CH3NGYZ/small-tailscale-openwrt #2: 通知申请支持bark

gzxiexl, 2025-04-17:
iPhone上面Bark用的最舒服了

CH3NGYZ, 2025-04-18:
现在可以再试试bark通知

gzxiexl, 2025-04-18:
还是无效
麻烦看下 bark的是
https://api.day.app/key/推送标题/这里改成你自己的推送内容

CH3NGYZ, 2025-04-18:
> 还是无效 麻烦看下 bark的是 [https://api.day.app/key/推送标题/这里改成你自己的推送内容](https://api.day.app/key/%E6%8E%A8%E9%80%81%E6%A0%87%E9%A2%98/%E8%BF%99%E9%87%8C%E6%94%B9%E6%88%90%E4%BD%A0%E8%87%AA%E5%B7%B1%E7%9A%84%E6%8E%A8%E9%80%81%E5%86%85%E5%AE%B9)

我修改了, 你试试更新本地版本, 脚本版本为1.0.3

gzxiexl, 2025-04-18:
我在试下 再建议下脚本包的版本号能不能显示在主菜单里

gzxiexl, 2025-04-18:
bark还是无效

CH3NGYZ, 2025-04-18:
> 我在试下 再建议下脚本包的版本号能不能显示在主菜单里

是显示在管理菜单的

CH3NGYZ, 2025-04-18:
稍等下我取个快递装下显卡, 待会做一下适配
[image]

gzxiexl, 2025-04-18:
重新跑了下 之前可能是老版本 界面更简陋

gzxiexl, 2025-04-18:
1.82.5貌似有点问题 你升级了没 我这边好像跑不起来 命令半天没反应

CH3NGYZ, 2025-04-18:
> 1.82.5貌似有点问题 你升级了没 我这边好像跑不起来 命令半天没反应

你如果没卸载就更新或者重装的话,运行tailscale up,会使用之前的配置, 不需要再次登录的

CH3NGYZ, 2025-04-18:
[image]

[image]

gzxiexl, 2025-04-18:
是的 没问题了  但是bark还是无效

CH3NGYZ, 2025-04-18:
> 是的 没问题了 但是bark还是无效

稍等, 我现在就弄, 因为我跟前没有iphone所以测试比较困难

CH3NGYZ, 2025-04-18:
你再试试
更新一下1.0.5版本

gzxiexl, 2025-04-18:
[image]
实际上还是没收到

CH3NGYZ, 2025-04-18:
bark地址应该是类似 `https://api.day/xxxxxxxx` 的形式

gzxiexl, 2025-04-18:
> bark key应该是类似 `https://api.day/xxxxxxxx` 的形式

没错呀 我确定没有配置错

CH3NGYZ, 2025-04-18:
试下1.0.8版本

gzxiexl, 2025-04-18:
https://api.day.app/mykey/
 Bark的是这个形式

CH3NGYZ, 2025-04-18:
> https://api.day.app/mykey/ Bark的是这个形式

是, 我是根据你前面输入的key, 和title等一起整合来的

```
if [ "$NOTIFY_BARK" = "1" ] && [ -n "$BARK_KEY" ]; then
        title_enc=$title
        content_enc="$content
        url="${BARK_KEY}/${title_enc}/${content_enc}"
        # 直接 GET 访问拼接好的地址，不带参数和 header
        if command -v curl > /dev/null; then
            curl -sS "$url" && echo "✅ Bark 通知已发送"
        elif command -v wget > /dev/null; then
            wget --quiet "$url" && echo "✅ Bark 通知已发送"
        else
            echo "❌ curl 和 wget 都不可用，无法发送 Bark 通知"
        fi
    fi
```

gzxiexl, 2025-04-18:
[image]
1.08还是报错

gzxiexl, 2025-04-18:
我家里和单位 两台openwrt上都是这样

gzxiexl, 2025-04-18:
```
# 检查是否需要发送 Bark 通知
if [[ "$NOTIFY_BARK" == "1" && -n "$BARK_KEY" ]]; then
    # 对标题和内容进行 URL 编码
    title_enc=$(urlencode "$title")
    content_enc=$(urlencode "$content")
    
    # 构建完整的 Bark 通知 URL
    url="https://api.day.app/${BARK_KEY}/${title_enc}/${content_enc}"
    
    # 使用 curl 发送请求
    if command -v curl > /dev/null; then
        response=$(curl -sS "$url")
        if [[ $? -eq 0 ]]; then
            echo "✅ Bark 通知已发送"
        else
            echo "❌ 发送 Bark 通知失败，HTTP 状态码: $response"
        fi
    elif command -v wget > /dev/null; then
        # 使用 wget 发送请求
        if wget --quiet --output-document=/dev/null "$url"; then
            echo "✅ Bark 通知已发送"
        else
            echo "❌ 发送 Bark 通知失败，wget 返回错误"
        fi
    else
        echo "❌ curl 和 wget 都不可用，无法发送 Bark 通知"
    fi
fi

```

urlencode 函数（需添加到脚本中）：
```
urlencode() {
    local length="${#1}"
    for (( i=0; i<length; i++ )); do
        local c="${1:i:1}"
        case "$c" in
            [a-zA-Z0-9._-]) 
                printf "$c"
                ;;
            *)
                printf "%%%02X" "'$c"
                ;;
        esac
    done
}

```
能试试吗？

gzxiexl, 2025-04-18:
[image]
版本号更后 提示更新

CH3NGYZ, 2025-04-18:
[image]

我这边 按9 更新完就没提示了

gzxiexl, 2025-04-18:
我下载了1.0.14版本的 还是bark无效 我在本地改了下tools.sh 试了一下 可以正常用bark了

gzxiexl, 2025-04-18:
[image]

CH3NGYZ, 2025-04-18:
> 我下载了1.0.14版本的 还是bark无效 我在本地改了下tools.sh 试了一下 可以正常用bark了

你发下你的tools.sh

gzxiexl, 2025-04-18:
bark的key是不带https://api.day.app/

gzxiexl, 2025-04-18:
[tools.zip](https://github.com/user-attachments/files/19808974/tools.zip)

CH3NGYZ, 2025-04-18:
因为存在自建服务器, 你的key还是加上前缀

gzxiexl, 2025-04-18:
上班去了 等下修改好了再更新

CH3NGYZ, 2025-04-18:
我还在测试  2种安装方式  与  是否自动更新  以及重启后的情况, 目前发现还是有点小bug

closed 2025-04-18 by gzxiexl