CH3NGYZ/small-tailscale-openwrt #3: bc: not found

gzxiexl, 2025-04-18:
```
root@ImmortalWrt:~# tailscale-helper 
[2025-04-18 15:14:19] 🔧 INFO: 🔄  正在检测脚本更新 ...
[2025-04-18 15:14:20] 🔧 INFO: 🎉  欢迎使用 Tailscale on OpenWRT 管理脚本 v1.0.18
[2025-04-18 15:14:20] 🔧 INFO: 📦  远程脚本版本: v1.0.10 🚨(有更新, 请按 [9] 更新)
[2025-04-18 15:14:20] 🔧 INFO:     请选择操作：
[2025-04-18 15:14:20] 🔧 INFO: 1)  📥 安装 / 重装 Tailscale
[2025-04-18 15:14:20] 🔧 INFO: 2)  🚀 启动 Tailscale
[2025-04-18 15:14:20] 🔧 INFO: 3)  🔄 管理 Tailscale 自动更新
[2025-04-18 15:14:20] 🔧 INFO: 4)  📦 查看本地 Tailscale 存在版本
[2025-04-18 15:14:20] 🔧 INFO: 5)  📦 查看远程 Tailscale 最新版本
[2025-04-18 15:14:20] 🔧 INFO: 6)  🔔 管理推送通知
[2025-04-18 15:14:20] 🔧 INFO: 7)  📊 排序代理池
[2025-04-18 15:14:20] 🔧 INFO: 8)  ♻️ 更新代理池
[2025-04-18 15:14:20] 🔧 INFO: 9)  🛠️ 更新脚本包
[2025-04-18 15:14:20] 🔧 INFO: 10) ❌ 卸载 Tailscale
[2025-04-18 15:14:20] 🔧 INFO: 0)  🚪 退出
[2025-04-18 15:14:20] 🔧 INFO: ✅ 请输入你的选择:
9
⬇️  尝试镜像: https://gh-proxy.com/github.com/CH3NGYZ/small-tailscale-openwrt/raw/refs/heads/main/tailscale-openwrt-scripts.tar.gz
❌ 校验失败！预期: a92ac2027aa062dceaec37775ec29b1c61e42c2ea7cae416ce7095c4dd0b9838，实际: 486a62ad38c21c1e1f16e6e1b5d902bc41d24963037b23a79d92cd454084b93c
⚠️ SHA256校验失败，尝试下一个镜像
❌ 校验失败！预期: 569194ff3cbb23333ec753078279ef5a，实际: bb5f8b0100badf5ce833049bf3f30778
⚠️ MD5校验失败，尝试下一个镜像
⬇️  尝试镜像: https://wget.la/https://github.com/CH3NGYZ/small-tailscale-openwrt/raw/refs/heads/main/tailscale-openwrt-scripts.tar.gz
❌ 校验失败！预期: a92ac2027aa062dceaec37775ec29b1c61e42c2ea7cae416ce7095c4dd0b9838，实际: 486a62ad38c21c1e1f16e6e1b5d902bc41d24963037b23a79d92cd454084b93c
⚠️ SHA256校验失败，尝试下一个镜像
❌ 校验失败！预期: 569194ff3cbb23333ec753078279ef5a，实际: bb5f8b0100badf5ce833049bf3f30778
⚠️ MD5校验失败，尝试下一个镜像
⬇️  尝试镜像: https://ghfast.top/https://github.com/CH3NGYZ/small-tailscale-openwrt/raw/refs/heads/main/tailscale-openwrt-scripts.tar.gz
❌ 校验失败！预期: a92ac2027aa062dceaec37775ec29b1c61e42c2ea7cae416ce7095c4dd0b9838，实际: 486a62ad38c21c1e1f16e6e1b5d902bc41d24963037b23a79d92cd454084b93c
⚠️ SHA256校验失败，尝试下一个镜像
❌ 校验失败！预期: 569194ff3cbb23333ec753078279ef5a，实际: bb5f8b0100badf5ce833049bf3f30778
⚠️ MD5校验失败，尝试下一个镜像
⬇️  尝试镜像: https://gh.catmak.name/https://github.com/CH3NGYZ/small-tailscale-openwrt/raw/refs/heads/main/tailscale-openwrt-scripts.tar.gz
❌ 校验失败！预期: a92ac2027aa062dceaec37775ec29b1c61e42c2ea7cae416ce7095c4dd0b9838，实际: 486a62ad38c21c1e1f16e6e1b5d902bc41d24963037b23a79d92cd454084b93c
⚠️ SHA256校验失败，尝试下一个镜像
❌ 校验失败！预期: 569194ff3cbb23333ec753078279ef5a，实际: bb5f8b0100badf5ce833049bf3f30778
⚠️ MD5校验失败，尝试下一个镜像
⬇️  尝试直连: CH3NGYZ/small-tailscale-openwrt/raw/refs/heads/main/tailscale-openwrt-scripts.tar.gz
❌ 所有镜像与直连均失败，安装中止
当前可用镜像地址列表 /etc/tailscale/valid_mirrors.txt 为:
https://gh-proxy.com/github.com/
https://wget.la/https://github.com/
https://ghfast.top/https://github.com/
https://gh.catmak.name/https://github.com/
您可能需要运行 /etc/tailscale/test_mirrors.sh 更新代理地址
[2025-04-18 15:14:31] 🔧 INFO: ✅ 脚本更新完毕, 请按回车重新加载...
[2025-04-18 15:14:39] 🔧 INFO: 🔄  正在检测脚本更新 ...
[2025-04-18 15:14:40] 🔧 INFO: 🎉  欢迎使用 Tailscale on OpenWRT 管理脚本 v1.0.18
[2025-04-18 15:14:40] 🔧 INFO: 📦  远程脚本版本: v1.0.10 🚨(有更新, 请按 [9] 更新)
[2025-04-18 15:14:40] 🔧 INFO:     请选择操作：
[2025-04-18 15:14:40] 🔧 INFO: 1)  📥 安装 / 重装 Tailscale
[2025-04-18 15:14:40] 🔧 INFO: 2)  🚀 启动 Tailscale
[2025-04-18 15:14:40] 🔧 INFO: 3)  🔄 管理 Tailscale 自动更新
[2025-04-18 15:14:40] 🔧 INFO: 4)  📦 查看本地 Tailscale 存在版本
[2025-04-18 15:14:40] 🔧 INFO: 5)  📦 查看远程 Tailscale 最新版本
[2025-04-18 15:14:40] 🔧 INFO: 6)  🔔 管理推送通知
[2025-04-18 15:14:40] 🔧 INFO: 7)  📊 排序代理池
[2025-04-18 15:14:40] 🔧 INFO: 8)  ♻️ 更新代理池
[2025-04-18 15:14:40] 🔧 INFO: 9)  🛠️ 更新脚本包
[2025-04-18 15:14:40] 🔧 INFO: 10) ❌ 卸载 Tailscale
[2025-04-18 15:14:40] 🔧 INFO: 0)  🚪 退出
[2025-04-18 15:14:40] 🔧 INFO: ✅ 请输入你的选择:
^C
root@ImmortalWrt:~# /etc/tailscale/test_mirrors.sh
测试 https://wget.la/https://github.com/ ...
/etc/tailscale/test_mirrors.sh: line 27: bc: not found
sh: invalid number ''
root@ImmortalWrt:~# /etc/tailscale/test_mirrors.sh
测试 https://wget.la/https://github.com/ ...
/etc/tailscale/test_mirrors.sh: line 27: bc: not found
sh: invalid number ''
root@ImmortalWrt:~# bc
-bash: bc: command not found
root@ImmortalWrt:~# 
```

可否换成awk
```
root@ImmortalWrt:~# awk
BusyBox v1.33.2 (2023-11-11 14:32:43 UTC) multi-call binary.

Usage: awk [OPTIONS] [AWK_PROGRAM] [FILE]...

        -v VAR=VAL      Set variable
        -F SEP          Use SEP as field separator
        -f FILE         Read program from FILE
        -e AWK_PROGRAM
root@ImmortalWrt:~# 
```

CH3NGYZ, 2025-04-18:
忘记换这个了, 我把pretest_mirrors里的换过来就行了

CH3NGYZ, 2025-04-18:
你测试一下能安装不

gzxiexl, 2025-04-19:
[image]
发现一个报错

CH3NGYZ, 2025-04-19:
nohup无法使用, 你是什么系统

gzxiexl, 2025-04-19:
istoreos

CH3NGYZ, 2025-04-19:
> istoreos

现在试试还行吗

gzxiexl, 2025-04-19:
[image]

gzxiexl, 2025-04-19:
360T7上装的immortalwrt就没这个报错

CH3NGYZ, 2025-04-21:
> 360T7上装的immortalwrt就没这个报错

我脚本里不是判断了如果没有nohup就用其他的命令吗，没生效吗

CH3NGYZ, 2025-04-21:
> 360T7上装的immortalwrt就没这个报错

是本地安装出现的这个问题吧？我之前设定的是后台更新，如果改成启动tailscale后运行更新脚本，阻塞开机启动脚本也不是不可以，因为tailscale已经启动了。

closed 2025-04-25 by CH3NGYZ