CH3NGYZ/small-tailscale-openwrt #4: 下载管理工具 & 排序代理池步骤，奇怪的 timeout

changhexiong, 2025-04-24:
```
root@router:~# [ "$(dtool)" = c ] && curl -fsSL $URL | sh || wget -qO- $URL | sh
[2025-04-24 18:56:58] [INFO] 🔗  使用固定代理下载: https://ghproxy.ch3ng.top/https://github.com/CH3NGYZ/small-tailscale-openwrt/raw/refs/heads/main/tailscale-openwrt-scripts.tar.gz
sh: timeout: not found
[2025-04-24 18:56:58] [INFO] 🔗  代理失效, 尝试直连: https://github.com/CH3NGYZ/small-tailscale-openwrt/raw/refs/heads/main/tailscale-openwrt-scripts.tar.gz
sh: timeout: not found
[2025-04-24 18:56:58] [ERROR] ❌  镜像与直连均失败, 安装中止
```

在同一设备上使用wget下载同一文件并无问题
```
root@router:~# wget -O test https://github.com/CH3NGYZ/small-tailscale-openwrt/raw/refs/heads/main/tailscale-openwrt-scripts.tar.gz
Downloading 'https://github.com/CH3NGYZ/small-tailscale-openwrt/raw/refs/heads/main/tailscale-openwrt-scripts.tar.gz'
Connecting to 20.205.243.166:443
Redirected to /CH3NGYZ/small-tailscale-openwrt/refs/heads/main/tailscale-openwrt-scripts.tar.gz on raw.githubusercontent.com
Writing to 'test'
test                 100% |*******************************| 14555   0:00:00 ETA
Download completed (14555 bytes)
```

系统是 Immortalwrt 24.10.1

CH3NGYZ, 2025-04-24:
`opkg install timeout`

你的timeout包没装，这是一个限定命令运行最大时长的包

CH3NGYZ, 2025-04-25:
如果上面的包安装失败或者找不到包, 试试 `opkg install coreutils` 后再试

closed 2025-04-25 by CH3NGYZ