CH3NGYZ/small-tailscale-openwrt #6: 下载或测速失败, 无法继续!

NJYZ-96, 2025-04-25:
日志如下

```
root@Z7-iStoreOS-alpha:~# mkdir /etc/tailscale
dtool(){ command -v curl >/dev/null&&echo c||command -v wget >/dev/null&&echo w||exit 1; }
root@Z7-iStoreOS-alpha:~# dtool(){ command -v curl >/dev/null&&echo c||command -v wget >/de
v/null&&echo w||exit 1; }
root@Z7-iStoreOS-alpha:~# URL="https://ghproxy.ch3ng.top/https://github.com/CH3NGYZ/small-t
ailscale-openwrt/raw/refs/heads/main/install.sh"
root@Z7-iStoreOS-alpha:~# [ "$(dtool)" = c ] && curl -fsSL $URL | sh || wget -qO- $URL | sh
[2025-04-25 15:45:48] [INFO] 📦  开始检查并安装必要组件...
[2025-04-25 15:45:48] [INFO] 🔄  正在更新 opkg 源...
[2025-04-25 15:45:53] [INFO] ✅  包 libustream-openssl 已安装，跳过
[2025-04-25 15:45:53] [INFO] ✅  包 ca-bundle 已安装，跳过
[2025-04-25 15:45:53] [INFO] ✅  包 kmod-tun 已安装，跳过
[2025-04-25 15:45:53] [INFO] ⚠️  包 coreutils-timeout 未安装，开始安装...
[2025-04-25 15:45:54] [INFO] ✅  包 coreutils-timeout 安装成功
[2025-04-25 15:45:54] [INFO] ⚠️  包 coreutils-nohup 未安装，开始安装...
[2025-04-25 15:45:54] [INFO] ✅  包 coreutils-nohup 安装成功
[2025-04-25 15:45:54] [INFO] ✅  timeout 命令已可用
[2025-04-25 15:45:54] [INFO] ✅  nohup 命令已可用
[2025-04-25 15:45:54] [INFO] 🔗  使用固定代理下载: https://ghproxy.ch3ng.top/https://github.com/CH3NGYZ/small-tailscale-openwrt/raw/refs/heads/main/tailscale-openwrt-scripts.tar.gz
[2025-04-25 15:45:55] [INFO] ✅  校验通过
[2025-04-25 15:45:55] [INFO] 📦  解压脚本包...
[2025-04-25 15:45:55] [INFO] ✅  软连接已成功创建：/etc/tailscale/helper.sh -> /usr/bin/tailscale-helper
[2025-04-25 15:45:55] [INFO] 🔍 本地不存在 mirrors.txt, 将下载镜像列表并测速, 请等待...
[2025-04-25 15:45:55] [INFO] 🔄  下载 pretest_mirrors.sh 并执行测速...
sh: can't open '/tmp/pretest_mirrors.sh': No such file or directory
[2025-04-25 15:45:56] [ERROR] ❌  下载或测速失败, 无法继续!

```

CH3NGYZ, 2025-04-25:
日志里提示下载失败了, 我看了下代码, 是一个变量名忘记修改了, 已更正, 请重试.

closed 2025-04-25 by CH3NGYZ

NJYZ-96, 2025-04-25:
> 日志里提示下载失败了, 我看了下代码, 是一个变量名忘记修改了, 已更正, 请重试.

还是有问题

```
 一键安装 Tailscale 配置工具安装完毕!
root@Z7-iStoreOS-alpha:~# tailscale-helper
[2025-04-25 16:45:33] [INFO] 🔄  正在检测脚本更新, 最多需要 10 秒...
[2025-04-25 16:45:34] [INFO] 🎉  欢迎使用 Tailscale on OpenWRT 管理脚本 v1.0.69
[2025-04-25 16:45:34] [INFO] 📦  远程脚本版本: v1.0.69 ✅已是最新
[2025-04-25 16:45:34] [INFO] ------------------------------------------
[2025-04-25 16:45:34] [INFO]       1).  💾 安装 / 重装 Tailscale
[2025-04-25 16:45:34] [INFO] ------------------------------------------
[2025-04-25 16:45:34] [INFO]       2).  📥 登录 Tailscale
[2025-04-25 16:45:34] [INFO]       3).  📝 生成 Tailscale 启动命令
[2025-04-25 16:45:34] [INFO]       4).  📤 登出 Tailscale
[2025-04-25 16:45:34] [INFO]       5).  ❌ 卸载 Tailscale
[2025-04-25 16:45:34] [INFO] ------------------------------------------
[2025-04-25 16:45:34] [INFO]       6).  🔄 管理 Tailscale 自动更新
[2025-04-25 16:45:34] [INFO]       7).  📦 查看本地 Tailscale 存在版本
[2025-04-25 16:45:34] [INFO]       8).  📦 查看远程 Tailscale 最新版本
[2025-04-25 16:45:34] [INFO]       9).  🔔 管理推送通知
[2025-04-25 16:45:34] [INFO]      10).  📊 排序代理池
[2025-04-25 16:45:34] [INFO]      11).  ♻️ 更新代理池
[2025-04-25 16:45:34] [INFO]      12).  🛠️ 更新脚本包
[2025-04-25 16:45:34] [INFO]      13).  📜 显示 Tailscale 更新日志
[2025-04-25 16:45:34] [INFO]      14).  🔄 手动运行更新脚本
[2025-04-25 16:45:34] [INFO] ------------------------------------------
[2025-04-25 16:45:34] [INFO]       0).  ⛔ 退出
[2025-04-25 16:45:34] [INFO] ------------------------------------------
[2025-04-25 16:45:34] [INFO] ✅  请输入你的选择: 1
[2025-04-25 16:45:37] [INFO]
[2025-04-25 16:45:37] [INFO] 加载公共函数...
[2025-04-25 16:45:37] [INFO] 加载配置文件...
[2025-04-25 16:45:37] [INFO]
[2025-04-25 16:45:37] [INFO] 📮 请选择安装模式：
[2025-04-25 16:45:37] [INFO]      1). 本地安装 (默认) 🏠
[2025-04-25 16:45:37] [INFO]      2). 内存安装 (临时) 💻
[2025-04-25 16:45:37] [INFO]      3). 退出           ⛔
[2025-04-25 16:45:37] [INFO] ⏳  请输入选项 [1/2/3]: 1
[2025-04-25 16:45:39] [INFO]
[2025-04-25 16:45:39] [INFO] 🔄  是否启用自动更新？
[2025-04-25 16:45:39] [INFO]       1). 是 (默认) ✅
[2025-04-25 16:45:39] [INFO]       2). 否        ❌
[2025-04-25 16:45:39] [INFO]       3). 退出      ⛔
[2025-04-25 16:45:39] [INFO] ⏳  请输入选项 [1/2/3]: 1
[2025-04-25 16:45:43] [INFO] 🔧  可用版本列表：
[2025-04-25 16:45:43] [INFO]   [1] v1.82.5
[2025-04-25 16:45:43] [INFO]   [2] v1.82.0
[2025-04-25 16:45:43] [INFO]   [3] v1.80.3
[2025-04-25 16:45:43] [INFO] ⏳  请输入序号选择版本 (留空使用 latest): 1
[2025-04-25 16:45:45] [INFO] ✅  使用指定版本: v1.82.5

[2025-04-25 16:45:45] [INFO] 🎯  当前安装配置：
[2025-04-25 16:45:45] [INFO] 🎯  模式: local
[2025-04-25 16:45:45] [INFO] 🎯  更新: true
[2025-04-25 16:45:45] [INFO] 🎯  版本: v1.82.5
[2025-04-25 16:45:45] [INFO] 🎯  架构: amd64
[2025-04-25 16:45:45] [INFO] 🎯  昵称: Z7-iStoreOS-alpha

[2025-04-25 16:45:45] [WARN] ⚠️  未找到 tailscale 服务文件，跳过停止服务步骤
[2025-04-25 16:45:45] [INFO] 🧹  清理残留文件...
[2025-04-25 16:45:45] [INFO] 🗑️  删除本地安装的残留文件...
[2025-04-25 16:45:45] [INFO] 🚀  开始安装 Tailscale...
[2025-04-25 16:45:45] [INFO] 🔗  准备校验文件...
[2025-04-25 16:45:45] [INFO] 🔗  尝试直接连接...
[2025-04-25 16:45:45] [WARN] ⚠️  无法获取 SHA256 校验文件
[2025-04-25 16:45:45] [INFO] 🔗  尝试直接连接...
[2025-04-25 16:45:45] [WARN] ⚠️  无法获取 MD5 校验文件
[2025-04-25 16:45:45] [INFO] 🔗  正在下载 Tailscale v1.82.5 (amd64)...
[2025-04-25 16:45:45] [INFO] 🔗  尝试直接连接...
[2025-04-25 16:45:45] [WARN] ⚠️  SHA256 校验失败，尝试使用 MD5...
[2025-04-25 16:45:45] [INFO] 🔗  尝试直接连接...
[2025-04-25 16:45:45] [ERROR] ❌  校验失败，安装中止
```

CH3NGYZ, 2025-04-25:
你的网络环境有问题, 我设置的代理不支持国外IP访问, 看看你有没有开clash, 漏网之鱼是不是设置的global?