CH3NGYZ/small-tailscale-openwrt #7: 当前默认代理 block 一些梯子 IP

changhexiong, 2025-04-25:
目前脚本文件里使用的代理套的是 CF，会自动block可疑的访问
因此，梯子的节点往往会被 403

对于已经配置了梯子的设备，安装反而会各种失败，需要手动把所有代理改成源地址才能成功

[image]

NJYZ-96, 2025-04-25:
是的，如果开了代理，哪怕是主路由的透明代理，都无法使用脚本；

回车执行都没有任何输出；关闭代理就好了

```
root@Z7-iStoreOS-alpha:~# dtool(){ command -v curl >/dev/null&&echo c||command -v wget >/de
v/null&&echo w||exit 1; }
root@Z7-iStoreOS-alpha:~# URL="https://ghproxy.ch3ng.top/https://github.com/CH3NGYZ/small-t
ailscale-openwrt/raw/refs/heads/main/install.sh"
root@Z7-iStoreOS-alpha:~# [ "$(dtool)" = c ] && curl -fsSL $URL | sh || wget -qO- $URL | sh
root@Z7-iStoreOS-alpha:~#
```

changhexiong, 2025-04-25:
> 我block了所有国外的ip, 你可以尝试fork本仓库修改所有代理为原地址, 要么在你的配置文件里添加上 `-DOAMIN_SUFFIX,ch3ng.top,DIRECT`

目前有两个地方会卡住
一个是下载管理工具 & 排序代理池
另一个是脚本中run_pretest_mirrors函数

下载管理工具 & 排序代理池这里或许可以不改，毕竟shell操作删一下就行；
但建议在脚本中run_pretest_mirrors函数下检测代理是否可用，如果不可用，fallback到源再尝试一次

NJYZ-96, 2025-04-25:
> 我block了所有国外的ip, 你可以尝试fork本仓库修改所有代理为原地址, 要么在你的配置文件里添加上 `- DOAMIN_SUFFIX,ch3ng.top,DIRECT` 因为要照顾tmp启动的设备, 往往开机启动时即使启动顺序放在99, 也和clash默认的99重合, 极有可能下载失败, 所以脚本中没有提供是否切换为仅使用直连的功能, 仅仅提供了fallback到原地址的功能 (之前踩过这个坑)

或许给个报错提示会比较好，目前开了代理没有任何日志

CH3NGYZ, 2025-04-25:
> > 我block了所有国外的ip, 你可以尝试fork本仓库修改所有代理为原地址, 要么在你的配置文件里添加上 `- DOAMIN_SUFFIX,ch3ng.top,DIRECT` 因为要照顾tmp启动的设备, 往往开机启动时即使启动顺序放在99, 也和clash默认的99重合, 极有可能下载失败, 所以脚本中没有提供是否切换为仅使用直连的功能, 仅仅提供了fallback到原地址的功能 (之前踩过这个坑)
> 
> 或许给个报错提示会比较好，目前开了代理没有任何日志

> > 我block了所有国外的ip, 你可以尝试fork本仓库修改所有代理为原地址, 要么在你的配置文件里添加上 `-DOAMIN_SUFFIX,ch3ng.top,DIRECT`
> 
> 目前安装过程有两个地方会卡住 一个是下载管理工具 & 排序代理池 另一个是脚本中run_pretest_mirrors函数
> 
> 下载管理工具 & 排序代理池这里或许可以不改，毕竟shell操作删一下就行； 但建议在脚本中run_pretest_mirrors函数下检测代理是否可用，如果不可用，fallback到源再尝试一次

我接下来做个纯直连的功能

closed 2025-04-29 by CH3NGYZ