CH3NGYZ/small-tailscale-openwrt #8: up了之后就报错了

lvangtao, 2025-04-28:
[2025-04-28 16:14:44] [INFO] 🎉  欢迎使用 Tailscale on OpenWRT 管理脚本 v1.0.70
[2025-04-28 16:14:44] [INFO] 📦  远程脚本版本: v1.0.70 ✅已是最新
[2025-04-28 16:14:44] [INFO] ------------------------------------------
[2025-04-28 16:14:44] [INFO]       1).  💾 安装 / 重装 Tailscale
[2025-04-28 16:14:44] [INFO] ------------------------------------------
[2025-04-28 16:14:44] [INFO]       2).  📥 登录 Tailscale
[2025-04-28 16:14:44] [INFO]       3).  📝 生成 Tailscale 启动命令
[2025-04-28 16:14:44] [INFO]       4).  📤 登出 Tailscale
[2025-04-28 16:14:44] [INFO]       5).  ❌ 卸载 Tailscale
[2025-04-28 16:14:44] [INFO] ------------------------------------------
[2025-04-28 16:14:44] [INFO]       6).  🔄 管理 Tailscale 自动更新
[2025-04-28 16:14:44] [INFO]       7).  📦 查看本地 Tailscale 存在版本
[2025-04-28 16:14:44] [INFO]       8).  📦 查看远程 Tailscale 最新版本
[2025-04-28 16:14:44] [INFO]       9).  🔔 管理推送通知
[2025-04-28 16:14:44] [INFO]      10).  📊 排序代理池
[2025-04-28 16:14:44] [INFO]      11).  ♻️ 更新代理池
[2025-04-28 16:14:44] [INFO]      12).  🛠️ 更新脚本包
[2025-04-28 16:14:44] [INFO]      13).  📜 显示 Tailscale 更新日志
[2025-04-28 16:14:44] [INFO]      14).  🔄 手动运行更新脚本
[2025-04-28 16:14:44] [INFO]      15).  🔄 切换代理/直连状态
[2025-04-28 16:14:44] [INFO] ------------------------------------------
[2025-04-28 16:14:44] [INFO]       0).  ⛔ 退出
[2025-04-28 16:14:44] [INFO] ------------------------------------------
[2025-04-28 16:14:44] [INFO] ✅  请输入你的选择: 2
[2025-04-28 16:15:01] [INFO] 
[2025-04-28 16:15:01] [INFO] 🚀  执行 tailscale up，正在监控输出...
[2025-04-28 16:15:01] [INFO] ✅  tailscale up 执行完成：/usr/bin/tailscale-helper: line 91: tailscale: command not found
__TS_UP_DONE__
[2025-04-28 16:15:01] [ERROR] ⚠️  tailscale 未登录或状态异常
[2025-04-28 16:15:01] [INFO] ✅  请按回车继续...

lvangtao, 2025-04-28:
已解决，以为脚本会一并自动安装程序，没想到要进入手动安装

closed 2025-04-28 by lvangtao

CH3NGYZ, 2025-04-28:
👍🏻