CH3NGYZ/small-tailscale-openwrt #9: 从原来的脚本升级问题

lichao0223, 2025-04-29:
现在在用https://github.com/CH3NGYZ/tailscale-openwrt 的脚本，是否可以无缝升级到当前的脚本呢，怕会丢登录信息

CH3NGYZ, 2025-04-29:
你是远程升级吗？远程不能直接升级到新脚本，而且我也没有测试过，最好先用之前的卸载脚本卸载，再重新安装这个

CH3NGYZ, 2025-04-29:
理论上下载安装一键脚本包没有任何问题，运行安装脚本时，也会覆盖原来的/etc/init.d/tailscale，理论上使用本仓库脚本装好tailscale后，重启即可升级成功，因为没有改变tailscale的配置，所以并不会有什么问题，但是为了保险，还是等我晚上回家测试一下能否正常使用吧

lichao0223, 2025-04-30:
> 理论上下载安装一键脚本包没有任何问题，运行安装脚本时，也会覆盖原来的/etc/init.d/tailscale，理论上使用本仓库脚本装好tailscale后，重启即可升级成功，因为没有改变tailscale的配置，所以并不会有什么问题，但是为了保险，还是等我晚上回家测试一下能否正常使用吧

请问有做测试吗

CH3NGYZ, 2025-05-05:
> > 理论上下载安装一键脚本包没有任何问题，运行安装脚本时，也会覆盖原来的/etc/init.d/tailscale，理论上使用本仓库脚本装好tailscale后，重启即可升级成功，因为没有改变tailscale的配置，所以并不会有什么问题，但是为了保险，还是等我晚上回家测试一下能否正常使用吧
> 
> 请问有做测试吗

抱歉, 五一在外面玩, 我现在做一下测试

CH3NGYZ, 2025-05-05:
可以无缝更换, 更换本仓库脚本后后需要进行一次安装, 之后重启, 运行 `rm /usr/bin/tailscale_downloader` 删掉原来的脚本即可, 我是本地测试的, 没有通过tailscale远程测试

closed 2025-05-05 by CH3NGYZ