CH3NGYZ/small-tailscale-openwrt #10: 安装成功后执行tailscale-helper提示: -ash: tailscale-helper: not found

Mr-P-hainan, 2025-05-11:
系统版本：ImmortalWrt 21.02-SNAPSHOT, r0-02fff2a
路由器芯片：MT7981/flogic 820
日志如下：
```
root@ImmortalWrt:~# rm -rf /etc/tailscale
root@ImmortalWrt:~# dtool(){ command -v curl >/dev/null&&echo c||command -v wget >/dev/null&&echo w||exit 1; }
root@ImmortalWrt:~# URL="https://ghproxy.ch3ng.top/https://github.com/CH3NGYZ/small-tailscale-openwrt/raw/refs/heads/main/install.sh"
root@ImmortalWrt:~# rm -f /tmp/tailscale-use-direct
root@ImmortalWrt:~# [ "$(dtool)" = c ] && curl -fsSL $URL | sh || wget -qO- $URL | sh
[2025-05-11 13:05:55] [INFO] 📦  开始检查并安装必要组件...
[2025-05-11 13:05:55] [INFO] 🔄  正在更新 opkg 源...
[2025-05-11 13:06:00] [INFO] ✅  包 libustream-openssl 已安装，跳过
[2025-05-11 13:06:00] [INFO] ✅  包 ca-bundle 已安装，跳过
[2025-05-11 13:06:00] [INFO] ✅  包 kmod-tun 已安装，跳过
[2025-05-11 13:06:00] [INFO] ✅  包 coreutils-timeout 已安装，跳过
[2025-05-11 13:06:00] [INFO] ✅  包 coreutils-nohup 已安装，跳过
[2025-05-11 13:06:00] [INFO] ✅  timeout 命令已可用
[2025-05-11 13:06:00] [INFO] ✅  nohup 命令已可用
[2025-05-11 13:06:00] [INFO] 🔗  使用固定代理下载: https://ghproxy.ch3ng.top/https://github.com/CH3NGYZ/small-tailscale-openwrt/raw/refs/heads/main/tailscale-openwrt-scripts.tar.gz
[2025-05-11 13:06:01] [INFO] ✅  校验通过
[2025-05-11 13:06:01] [INFO] 📦  解压脚本包...
[2025-05-11 13:06:01] [INFO] ✅  软连接已成功创建：/etc/tailscale/helper.sh -> /usr/bin/tailscale-helper
[2025-05-11 13:06:01] [INFO] 🔍 本地不存在 mirrors.txt, 将下载镜像列表并测速, 请等待...
[2025-05-11 13:06:01] [INFO] 🔄  下载 pretest_mirrors.sh 并执行测速...
[2025-05-11 13:06:02] [INFO] 🔗  使用固定代理下载: https://ghproxy.ch3ng.top/https://github.com/CH3NGYZ/small-tailscale-openwrt/releases/download/v1.82.5/SHA256SUMS.txt
[2025-05-11 13:06:04] [INFO] 🛠️  正在下载镜像列表，请耐心等待...
[2025-05-11 13:06:05] [INFO] ✅  已更新镜像列表
[2025-05-11 13:06:05] [WARN] ⚠️  测试代理下载tailscale可执行文件花费的时间中, 每个代理最长需要 10 秒, 请耐心等待......
[2025-05-11 13:06:05] [INFO] ⏳   测试[1/12] https://wget.la/https://github.com/
[2025-05-11 13:06:08] [INFO] ✅  用时 3.00s
[2025-05-11 13:06:08] [INFO] ⏳   测试[2/12] https://fastgit.cc/https://github.com/
[2025-05-11 13:06:10] [INFO] ✅  用时 2.00s
[2025-05-11 13:06:10] [INFO] ⏳   测试[3/12] https://ghfast.top/https://github.com/
[2025-05-11 13:06:14] [INFO] ✅  用时 4.00s
[2025-05-11 13:06:14] [INFO] ⏳   测试[4/12] https://gh-proxy.com/github.com/
[2025-05-11 13:06:15] [INFO] ✅  用时 1.00s
[2025-05-11 13:06:15] [INFO] ⏳   测试[5/12] https://ghproxy.net/https://github.com/
[2025-05-11 13:06:16] [WARN] ❌  下载失败
[2025-05-11 13:06:16] [INFO] ⏳   测试[6/12] https://gh.catmak.name/https://github.com/
[2025-05-11 13:06:19] [INFO] ✅  用时 3.00s
[2025-05-11 13:06:19] [INFO] ⏳   测试[7/12] https://github.3x25.com/https://github.com/
[2025-05-11 13:06:22] [INFO] ✅  用时 3.00s
[2025-05-11 13:06:22] [INFO] ⏳   测试[8/12] https://mirrors.chenby.cn/https://github.com/
[2025-05-11 13:06:24] [INFO] ✅  用时 2.00s
[2025-05-11 13:06:24] [INFO] ⏳   测试[9/12] https://down.npee.cn/?https://github.com/
[2025-05-11 13:06:26] [INFO] ✅  用时 2.00s
[2025-05-11 13:06:26] [INFO] ⏳   测试[10/12] https://cdn.crashmc.com/https://github.com/
[2025-05-11 13:06:29] [INFO] ✅  用时 3.00s
[2025-05-11 13:06:29] [INFO] ⏳   测试[11/12] https://gh-proxy.ygxz.in/https://github.com/
[2025-05-11 13:06:32] [INFO] ✅  用时 3.00s
[2025-05-11 13:06:32] [INFO] ⏳   测试[12/12] https://ghproxy.cxkpro.top/https://github.com/
[2025-05-11 13:06:35] [INFO] ✅  用时 3.00s
[2025-05-11 13:06:35] [INFO] 🏆 最佳镜像: https://gh-proxy.com/github.com/
[2025-05-11 13:06:35] [INFO] ✅  下载镜像列表并测速完成！
[2025-05-11 13:06:35] [INFO] ✅  一键安装 Tailscale 配置工具安装完毕!
root@ImmortalWrt:~# tailscale-helper
-ash: tailscale-helper: not found
```

CH3NGYZ, 2025-05-11:
执行这个命令试试: `/usr/bin/tailscale-helper`

Mr-P-hainan, 2025-05-11:
不行，还是一样的结果：
```
root@ImmortalWrt:~# # 代理版
root@ImmortalWrt:~# rm -rf /etc/tailscale
root@ImmortalWrt:~# dtool(){ command -v curl >/dev/null&&echo c||command -v wget >/dev/null&&echo w||exit 1; }
root@ImmortalWrt:~# URL="https://ghproxy.ch3ng.top/https://github.com/CH3NGYZ/small-tailscale-openwrt/raw/refs/heads/main/install.sh"
root@ImmortalWrt:~# rm -f /tmp/tailscale-use-direct
root@ImmortalWrt:~# [ "$(dtool)" = c ] && curl -fsSL $URL | sh || wget -qO- $URL | sh
[2025-05-11 13:19:46] [INFO] 📦  开始检查并安装必要组件...
[2025-05-11 13:19:46] [INFO] 🔄  正在更新 opkg 源...
[2025-05-11 13:19:52] [INFO] ✅  包 libustream-openssl 已安装，跳过
[2025-05-11 13:19:52] [INFO] ✅  包 ca-bundle 已安装，跳过
[2025-05-11 13:19:52] [INFO] ✅  包 kmod-tun 已安装，跳过
[2025-05-11 13:19:52] [INFO] ✅  包 coreutils-timeout 已安装，跳过
[2025-05-11 13:19:52] [INFO] ✅  包 coreutils-nohup 已安装，跳过
[2025-05-11 13:19:52] [INFO] ✅  timeout 命令已可用
[2025-05-11 13:19:52] [INFO] ✅  nohup 命令已可用
[2025-05-11 13:19:52] [INFO] 🔗  使用固定代理下载: https://ghproxy.ch3ng.top/https://github.com/CH3NGYZ/small-tailscale-openwrt/raw/refs/heads/main/tailscale-openwrt-scripts.tar.gz
[2025-05-11 13:19:54] [INFO] ✅  校验通过
[2025-05-11 13:19:54] [INFO] 📦  解压脚本包...
[2025-05-11 13:19:54] [INFO] ✅  软连接已成功创建：/etc/tailscale/helper.sh -> /usr/bin/tailscale-helper
[2025-05-11 13:19:54] [INFO] 🔍 本地不存在 mirrors.txt, 将下载镜像列表并测速, 请等待...
[2025-05-11 13:19:54] [INFO] 🔄  下载 pretest_mirrors.sh 并执行测速...
[2025-05-11 13:19:55] [INFO] 🔗  使用固定代理下载: https://ghproxy.ch3ng.top/https://github.com/CH3NGYZ/small-tailscale-openwrt/releases/download/v1.82.5/SHA256SUMS.txt
[2025-05-11 13:19:56] [INFO] 🛠️  正在下载镜像列表，请耐心等待...
[2025-05-11 13:19:57] [INFO] ✅  已更新镜像列表
[2025-05-11 13:19:57] [WARN] ⚠️  测试代理下载tailscale可执行文件花费的时间中, 每个代理最长需要 10 秒, 请耐心等待......
[2025-05-11 13:19:57] [INFO] ⏳   测试[1/12] https://wget.la/https://github.com/
[2025-05-11 13:20:00] [INFO] ✅  用时 3.00s
[2025-05-11 13:20:00] [INFO] ⏳   测试[2/12] https://fastgit.cc/https://github.com/
[2025-05-11 13:20:03] [INFO] ✅  用时 3.00s
[2025-05-11 13:20:03] [INFO] ⏳   测试[3/12] https://ghfast.top/https://github.com/
[2025-05-11 13:20:06] [INFO] ✅  用时 3.00s
[2025-05-11 13:20:06] [INFO] ⏳   测试[4/12] https://gh-proxy.com/github.com/
[2025-05-11 13:20:07] [INFO] ✅  用时 1.00s
[2025-05-11 13:20:07] [INFO] ⏳   测试[5/12] https://ghproxy.net/https://github.com/
[2025-05-11 13:20:08] [WARN] ❌  下载失败
[2025-05-11 13:20:08] [INFO] ⏳   测试[6/12] https://gh.catmak.name/https://github.com/
[2025-05-11 13:20:11] [INFO] ✅  用时 3.00s
[2025-05-11 13:20:11] [INFO] ⏳   测试[7/12] https://github.3x25.com/https://github.com/
[2025-05-11 13:20:14] [INFO] ✅  用时 3.00s
[2025-05-11 13:20:14] [INFO] ⏳   测试[8/12] https://mirrors.chenby.cn/https://github.com/
[2025-05-11 13:20:17] [INFO] ✅  用时 3.00s
[2025-05-11 13:20:17] [INFO] ⏳   测试[9/12] https://down.npee.cn/?https://github.com/
[2025-05-11 13:20:20] [INFO] ✅  用时 3.00s
[2025-05-11 13:20:20] [INFO] ⏳   测试[10/12] https://cdn.crashmc.com/https://github.com/
[2025-05-11 13:20:23] [INFO] ✅  用时 3.00s
[2025-05-11 13:20:23] [INFO] ⏳   测试[11/12] https://gh-proxy.ygxz.in/https://github.com/
[2025-05-11 13:20:27] [INFO] ✅  用时 4.00s
[2025-05-11 13:20:27] [INFO] ⏳   测试[12/12] https://ghproxy.cxkpro.top/https://github.com/
[2025-05-11 13:20:30] [INFO] ✅  用时 3.00s
[2025-05-11 13:20:30] [INFO] 🏆 最佳镜像: https://gh-proxy.com/github.com/
[2025-05-11 13:20:30] [INFO] ✅  下载镜像列表并测速完成！
[2025-05-11 13:20:30] [INFO] ✅  一键安装 Tailscale 配置工具安装完毕!
root@ImmortalWrt:~# /usr/bin/tailscale-helper
-ash: /usr/bin/tailscale-helper: not found
```

CH3NGYZ, 2025-05-11:
```/etc/tailscale/helper.sh``` 这个呢

CH3NGYZ, 2025-05-11:
可是有这个判断, 不是文件找不到的问题
``` bash
if [ -L /usr/bin/tailscale-helper ]; then
    log_info "✅  软连接已成功创建：$CONFIG_DIR/helper.sh -> /usr/bin/tailscale-helper"
else
    log_error "❌  创建软连接失败"
fi
```

试试
``` bash
sh /usr/bin/tailscale-helper
```

Mr-P-hainan, 2025-05-11:
执行结果如下：
```
root@ImmortalWrt:~# sh /usr/bin/tailscale-helper
/usr/bin/tailscale-helper: line 102: syntax error: unexpected redirection
```

CH3NGYZ, 2025-05-11:
稍等, 我知道问题在哪了, 是因为你可能没有bash, 而只有sh, 脚本利用到了一些bash独有的语法, 所以我还得适配一下更通用的写法

``` bash
ls -l /bin/bash
```
若输出为 `No such file or directory`，那就说明确实没有 bash

Mr-P-hainan, 2025-05-11:
> 稍等, 我知道问题在哪了, 是因为你可能没有bash, 而只有sh, 脚本利用到了一些bash独有的语法, 所以我还得适配一下更通用的写法
> 
> ls -l /bin/bash
> 若输出为 `No such file or directory`，那就说明确实没有 bash

正确，我想着也是因为这个，我试着直接执行helper.sh但是提示了-ash: command not found
```
root@ImmortalWrt:~# ls -l /bin/bash
ls: /bin/bash: No such file or directory
```

Mr-P-hainan, 2025-05-11:
通过opkg安装bash后可以正常使用

closed 2025-05-11 by Mr-P-hainan

CH3NGYZ, 2025-05-11:
你其实可以试试最新版了, 我全都换成了 `#!/bin/sh`