CH3NGYZ/small-tailscale-openwrt #12: Support mips_siflower arch

inoribea, 2025-05-15:
like this 

https://github.com/lanrat/openwrt-tailscale-repo

https://lanrat.github.io/openwrt-tailscale-repo/packages/19.07/tailscale_1.82.5_mips_siflower.ipk

CH3NGYZ, 2025-05-16:
Based on [https://github.com/lanrat/openwrt-tailscale-repo/blob/main/build.sh#L15](https://github.com/lanrat/openwrt-tailscale-repo/blob/main/build.sh#L15), the architecture for mips\_siflower during compilation is mipsle, using hardfloat. I might need a few days to finish my work before I can start the adaptation work. Please understand.

CH3NGYZ, 2025-06-07:
Sorry, I'm too busy with work and may not have time to support this architecture. However, you're welcome to submit a PR with your code.

closed 2025-06-07 by CH3NGYZ