CH3NGYZ/small-tailscale-openwrt #17: 子网路由失效了，求分享一下启动参数

gzxiexl, 2025-09-19:
原来的启动参数如下
tailscale up --stateful-filtering=false --accept-routes=true --accept-dns=false --advertise-exit-node --advertise-routes=192.168.8.0/24 --reset

不知道从哪个版本开始 子网路由无效了

CH3NGYZ, 2025-09-21:
或许你可以检查一下后台日志, 看看tailscaled 服务运行正不正常

lichao0223, 2025-10-21:
Tue Oct 21 10:05:51 2025 daemon.err tailscaled[19043]: 2025/10/21 02:05:51 magicsock: closing connection to derp-2 (idle), age 1m0s
Tue Oct 21 10:05:51 2025 daemon.err tailscaled[19043]: 2025/10/21 02:05:51 magicsock: 1 active derp conns: derp-10=cr10m0s,wr10m0s
Tue Oct 21 10:05:51 2025 daemon.err tailscaled[19043]: panic: unreachable
Tue Oct 21 10:05:51 2025 daemon.err tailscaled[19043]:
Tue Oct 21 10:05:51 2025 daemon.err tailscaled[19043]: goroutine 219 [running]:
Tue Oct 21 10:05:51 2025 daemon.err tailscaled[19043]: tailscale.com/net/tstun.invertGSOChecksum(...)
Tue Oct 21 10:05:51 2025 daemon.err tailscaled[19043]:  tailscale.com/net/tstun/wrap.go:1047
Tue Oct 21 10:05:51 2025 daemon.err tailscaled[19043]: tailscale.com/net/tstun.(*Wrapper).injectedRead(0xc0001409a0?, {0x0?, {0xc003f7f260?, 0xc002dc5d30?, 0xc000083880?}}, {0xc00027e588?, 0xc000083880?, 0x4?}, {0xc00055cc00, 0x80, ...}, ...)
Tue Oct 21 10:05:51 2025 daemon.err tailscaled[19043]:  tailscale.com/net/tstun/wrap.go:1087 +0x159
Tue Oct 21 10:05:51 2025 daemon.err tailscaled[19043]: tailscale.com/net/tstun.(*Wrapper).Read(0xc0001409a0, {0xc00027e588, 0x80, 0x80}, {0xc00055cc00, 0x80, 0x80}, 0x18)
Tue Oct 21 10:05:51 2025 daemon.err tailscaled[19043]:  tailscale.com/net/tstun/wrap.go:938 +0x285
Tue Oct 21 10:05:51 2025 daemon.err tailscaled[19043]: github.com/tailscale/wireguard-go/device.(*Device).RoutineReadFromTUN(0xc000160288)
Tue Oct 21 10:05:51 2025 daemon.err tailscaled[19043]:  github.com/tailscale/wireguard-go@v0.0.0-20250716170648-1d0488a3d7da/device/send.go:249 +0x317
Tue Oct 21 10:05:51 2025 daemon.err tailscaled[19043]: created by github.com/tailscale/wireguard-go/device.NewDevice in goroutine 177
Tue Oct 21 10:05:51 2025 daemon.err tailscaled[19043]:  github.com/tailscale/wireguard-go@v0.0.0-20250716170648-1d0488a3d7da/device/device.go:322 +0x445 
像是 tailscale 本身问题，一访问子网就 panic

closed 2025-10-25 by CH3NGYZ

lichao0223, 2025-10-25:
> 由于最近tailscale官方仓库用于编译可执行文件的脚本 build_dist.sh 更新过，所以本仓库的编译脚本编译出来的文件有点问题，现已还原为通用编译方式，请尝试最新版本，如果问题依旧存在，请REOPEN ISSUE。
> 
> > 经过测试，还原到 1.86.2 后正常

libustream-openssl 未安装，开始安装...
[2025-10-25 10:28:53] [ERROR] ❌ 安装 libustream-openssl 失败，无法继续，请手动安装此包
[2025-10-25 10:28:53] [ERROR] ❌  脚本更新失败, 脚本内置作者的代理失效
对了，脚本有这个问题，最新的istoreos的SSL 支持库从 openssl 切换到 mbedtls，所以脚本里的libustream-openssl包检测必须去掉才能执行，这块能否考虑一下